Molodos/ElegooNeptuneThumbnails-Prusa #1: Elegoo Neptune 4 Pro Thumbnail does not appear.

CapciGithub, 2023-09-01:
Hello, I tried the code that you made, and I followed every instruction just like how you said but for some reason thumbnails does not appear on Elegoo Neptune 4 Pro screen nor the fluidd interface.

Here's my gcode;
[Vase_01_SMALL_0.4n_0.3mm_PLA_M5_2h44m.gcode.zip](https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa/files/12501023/Vase_01_SMALL_0.4n_0.3mm_PLA_M5_2h44m.gcode.zip)

Molodos, 2023-09-02:
Hi, it seems like you have a M5 selected as a printer. As my script detects the printer and this isn't an Elegoo printer, it is not working. Also, the script does not add fluidd thumbnails, so that part is not a bug.
For people using other printers in the config, I will most probably update the script and instructions to add an override. I'll notify you here as soon as there is news

CapciGithub, 2023-09-02:
> Hi, it seems like you have a M5 selected as a printer. As my script detects the printer and this isn't an Elegoo printer, it is not working. Also, the script does not add fluidd thumbnails, so that [...]

Oh, my bad. I'm new into this and It was some config that i found over reddit and It was for my Elegoo Neptune 4 Printer. Thanks for info though.

Molodos, 2023-09-02:
Update to set printer model manually is out. Download the latest release and read the new FAQ to check out how to do it.
For the fluidd thing: Check out that guide: https://docs.fluidd.xyz/features/thumbnails#prusaslicer

closed 2023-09-02 by Molodos

knkacodelazm, 2023-09-08:
KRAL SEN TURKMUSUN TURKSEN SU CID CREATORE EL ATMA GIBI BI SANSIN VARMI GERCEKTEN COK IHTIYACIM VAR KRAL

Molodos, 2023-09-08:
> KRAL SEN TURKMUSUN TURKSEN SU CID CREATORE EL ATMA GIBI BI SANSIN VARMI GERCEKTEN COK IHTIYACIM VAR KRAL

WTF? What about disabling caps and writing english. I might understand you then