Molodos/ElegooNeptuneThumbnails-Prusa #3: does not work with Firmware 1.1.2.11

Primus007, 2023-09-26:
i followed your instructions but does not work with Firmware 1.1.2.11 TFT 1.2.9

Molodos, 2023-09-26:
Sorry, but that's not enough info for me to help, I do not even know what printer you are using. Could you please provide the following Info:
- What printer are you using?
- The .gcode file that did not work
Thank you

Primus007, 2023-09-26:
sorry I forgot this infos to add.
My Printer is the Elegoo Neptune 4 (non pro)
The .gcode is a simple small test cube. 
[Shape-Box.zip](https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa/files/12727464/Shape-Box.zip)

Molodos, 2023-09-26:
Looks like you are not using a N4 printer preset and also didn't specify the printer model manually. My script needs one of them in order to know how the thumbnails should be formatted.
Check out the FAQ, it explains exactly what to do: https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa#i-did-not-use-the-official-neptune-printer-preset-in-prusasliceer-what-to-do

Primus007, 2023-09-26:
Thank your for the info with the presets. i added this --printer=NEPTUNE4 
but it still does not work.
i'm pretty sure something has been changed in the new firmware and that's why it doesn't work anymore.
because there is another thumbnail project on github. that always worked until i installed the new firmware. so i think with your project it is the same problem.

Molodos, 2023-09-26:
I just tested it with the newest PrusaSlicer version 2.6.1 and it worked just fine. You also send me the gcode where you added the "--printer=NEPTUNE4"? Maybe I can find something, there. Also you might wanna try placing the 'ElegooNeptuneThumbnails-Prusa.exe" into another location. I don't know if that would help, but it is worth a try :D

Primus007, 2023-09-26:
[Shape-Box1.zip](https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa/files/12729577/Shape-Box1.zip)

Your plugin makes changes in the gcode file, I see that. So it is doing its job.
i suspect that elegoo has changed something in the new firmware, so the thumbnail is no longer displayed. maybe the thumbnail format has been changed or something else.

Molodos, 2023-09-26:
Okay, now the gcode looks good. What I missed was, that you were talking about the new Neptune firmware and I was thinking about the new PrusaSlicer version. I will do some tests with my printer and then I will get back to you here :D

Molodos, 2023-09-26:
So my printers are still on the old firmware. Could you tell me where you got the update files for your N4 from?

Primus007, 2023-09-26:
i got the firmware from the discord channel: https://discord.com/channels/969282195552346202/1152157413089222676/1152489982615883846
1.1.2.11.zip
however, it was later recommended not to install this firmware because it is buggy. however, it works fine for me and for most others. 

i can confirm too, with the old firmware the thumbnail was displayed.

Primus007, 2023-09-26:
i will let you know as soon as i have tried it with the new firmware.

Molodos, 2023-09-27:
I got a response from Elegoo. They said, that they tried my plugin on the latest firmware version and it did work. Where exectly did the thumbnail not work? While printing or in the preview before hitting print? Because in the preview you might have to go back and open it again to show, that was always a bit buggy for me :)

Primus007, 2023-09-27:
In fluidd the thumbnail is visible but on the display not.
i am not alone with this problem. https://discord.com/channels/969282195552346202/1141576944702476368
it is about another plugin but it does the same as your plugin. 
two plugins that suddenly stop working. this is not a coincidence.

Molodos, 2023-09-27:
Okay, I see. But i have an idea. I think I will update one of my printers today (won't risk both) and try something. I'll keep you updated

Molodos, 2023-09-29:
Quick update: I am sorry, but I just realized, that I forgot updating and testing. As I now am away for a week or so, it will take some more time unfortunately. Just wanted to let you know (maybe there will even be a new update until then, because I saw many people recommending to not update just now)

Primus007, 2023-09-30:
No problem.
Currently i am back on FW 1.1.1.98 and your script works fine. it seems to be due to the firmware. let's wait and see if it works with the new firmware.

Molodos, 2023-11-21:
Hi, I have seen, that a new stable firmware version is out, now. Have you by any chance already tried it out in order to confirm if the issue is still there? :)

Primus007, 2023-11-21:
i am on stock klipper. and there is no issue with you tool.

Molodos, 2023-11-21:
I understand, in that case, I will close this issue :)

closed 2023-11-21 by Molodos