Molodos/ElegooNeptuneThumbnails-Prusa #5: does not work with Mac M-series processors

jKawka-ITV, 2023-10-21:
Trying to get it to run on a MacBook Pro M1. 

I've compiled it using instructions, but trying to run it it throws an error over ColPic_X64.dll being an x86 executable as opposed to the arm64 required which makes sense. 

Are there binaries anywhere so that I can compile them into a compatible lib?

jKawka-ITV, 2023-10-21:
Sorry my bad, it correctly identifies the .dylib file for Mac, but again needs an arm64 version.

[image]

Molodos, 2023-10-21:
Unfortunately I cannot provide any other binaries as I extracted them from Elegoo Cura, whicht provides only the binaries in this repository. As I do also not know how exactly they work internally and I do not have a mac, I am afraid, that I cannot help you there, I am sorry :/

Molodos, 2023-10-22:
Did you try out Elegoo Cura already? Because if thumbnails work with it, there might be a chance for getting it to work here as well :)

Molodos, 2023-10-22:
Another user checked Elegoo Cura for me and came to the result, that it also doesn't work. Unfortunately I am dependent of the official Elegoo Cura binaries and as there are no binaries for the MacOS M series Chips, I am not able to provide support for those.

Molodos, 2023-11-06:
Reopening the issue as I am currently reverse engineering the Elegoo binaries with the goal of re-writing them in python to get rid of those dependencies that cause errors. Could take some more time, but just wanted to keep you updated :)

Anvilondre, 2023-12-23:
@jKawka-ITV can you try the instructions in my PR?

Molodos, 2023-12-23:
Good news! Version 2.0.0 of the post processing script is out. The core was reworked in order to remove dependencies of external libraries. This should fix all the M series chip problems. (compile instructions were also updated)
I am happy to hear your feedback if everything works as I do not have a mac to test on :D

ruiaraujo012, 2023-12-23:
I have some problem trying to compile on my mac M1 
[image]

ruiaraujo012, 2023-12-23:
Ok it worked with `python -m PyInstaller --onefile --name="ElegooNeptuneThumbnails-Prusa" elegoo_neptune_thumbnails.py`

ruiaraujo012, 2023-12-23:
But it throws an error generating the gcode 
```
Documents/scripts/OrcaSlicerThumbnail/ElegooNeptuneThumbnails-Prusa/dist/ElegooN
eptuneThumbnails-Prusa --printer=NEPTUNE4PRO on file
/var/folders/3j/d8hbvkfs5wl1_r15xnn1tzkc0000gn/T/orcaslicer_model/Sat_Dec_23/19_25_27#21389#53/
Metadata/.21389.0.gcode.pp failed.
Error code: 1
Output:
Traceback (most recent call last):
File "elegoo_neptune_thumbnails.py", line 12, in <module>
File "PyInstaller/loader/pyimod02_importers.py", line 419, in exec_module
File "PyQt6/__init__.py", line 20, in <module>
ModuleNotFoundError: No module named 'pkgutil'
[42460] Failed to execute script 'elegoo_neptune_thumbnails' due to unhandled exception!
```

Molodos, 2023-12-23:
That's weird. Maybe you could try the instructions for the PR (https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa/pull/13/files) of @Anvilondre to build inside the right environment.

ruiaraujo012, 2023-12-23:
It worked perfectly with those instructions, thanks.

Molodos, 2023-12-23:
Nice, then I guess I will merge it :D

Molodos, 2023-12-23:
Merged, closing this issue as resolved

closed 2023-12-23 by Molodos