Molodos/ElegooNeptuneThumbnails-Prusa #6: Default thumbnail size in orca makes the exe crash

heolfief, 2023-10-28:
Hello,
I'm using Orca slicer 1.7.0 with elegoo Neptune 4 pro.
The default thumbnail size in Orca for this printer is 300x300, which makes this program crash.

```powershell
Traceback (most recent call last):
  File "elegoo_neptune_thumbnails.py", line 248, in <module>
  File "elegoo_neptune_thumbnails.py", line 25, in __init__
  File "elegoo_neptune_thumbnails.py", line 69, in _get_q_image_thumbnail
  File "elegoo_neptune_thumbnails.py", line 62, in _get_base64_thumbnail
Exception: Thumbnail is not present
[5576] Failed to execute script 'elegoo_neptune_thumbnails' due to unhandled exception!
``` 

It seems to be due to the script looking for a specific size of 600x600 and not handling other sizes.
Code itself is here: https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa/blob/87b2c4691ffa25245293e786c036ba2cfe5a5463/elegoo_neptune_thumbnails.py#L54C66-L54C66

Is this something that could be improved ? I had to execute the script from a command line to get this error log, then dig in the python code to understand that I should set 600x600 in my slicer settings

Thanks

Molodos, 2023-10-28:
As the Readme already says, a size of 600x600 is required for the script to work. This could be changed, but it would make thinks more complicated and the result would be, that scaling the image up would reduce quality of the displayed image on the printer. Therefore I do not plan to add such a thing.
Nevertheless you can just set the Thumbnail size of Orca to 600x600 and it will work. Also you can set multiple sizes if you for some reason need the 300x300 one, too.
Or is there something, that prevents you from changing the size to easily fix the problem?

heolfief, 2023-10-28:
Right, I can understand that changing the size to something else than 600 is a big change, and maybe not worth.
I meant that the error handling itself could be improved, in this case, just a popup saying that I need to make sure the thumbnail is 600x600 would be a great addition, especially because the setting itself in Orca is not at the same place as in Prusa, so following the install readme directly is not always "easy".
Thanks 🙂

Molodos, 2023-10-28:
You're totally right. I just quickly improved it and published a new release :)

closed 2023-10-28 by Molodos