Molodos/ElegooNeptuneThumbnails-Prusa #8: Neptune 4 Max

Kiroorik, 2023-11-06:
Please add the Neptune 4 Max profile

Molodos, 2023-11-06:
I definitely plan adding support for those printers if I can somehow get access to them. I am currently already in contact with Elegoo regarding that :)

Goodsmileduck, 2023-11-08:
let me know if you need help with testing or debugging, I have one Neptune 4 Max.
Also see people just using `--printer=NEPTUNE4PRO` seems working for them (found on reddit)
I will try it tomorrow

Goodsmileduck, 2023-11-09:
It actually works, it's showing thumbnail on screen!

Molodos, 2023-11-21:
Hi, sorry for the late reply. Ad [Goodsmileduck](https://github.com/Goodsmileduck) already said, you can try using --printer=NEPTUNE4PRO, which most likely works. It would be nice if someone could confirm that for me (I know, that Goodsmileduck also sent an image, but it seems a bit weird to be as the background is black and does not have the grid as seen in the Readme, so further confirmation would be nice)
After confirmation, I will officially add a Max and Plus option :)

HautdogWater, 2023-11-25:
I can confirm this works for the Neptune 4 Plus as well.

Molodos, 2023-11-26:
> I can confirm this works for the Neptune 4 Plus as well.

Thanks for that info, I will use it to officially support the N4Plus soon

Molodos, 2023-11-28:
Good news: The Neptune 4 Plus and Max are now officially supported

closed 2023-11-28 by Molodos