Molodos/ElegooNeptuneThumbnails-Prusa #9: Issue on Neptune 4

DNicholai, 2023-11-14:
Firstly - great software. It works great, from the command line :) 

I'm having an issue on a linux based install with a Neptune 4.  

Script runs fine when I run it manually in the command line, but when I run it from PrusaSlicer, I get the following error : 

_Post-processing script /home/dpegon/SRC/ElegooNeptuneThumbnails-Prusa-1.1.5/dist/ElegooNeptuneThumbnails-Prusa --printer=NEPTUNE4PRO on file /tmp/.PrusaSlicer.upload.8c4e-31dd-d5ae-e39b failed.
Error code: 255
Output:
[104] Error loading Python lib '/tmp/_MEIWjUxOl/libpython3.10.so.1.0': dlopen:_

I've tried with and without the post command append (--printer=NEPTUNE4PRO)

Any hints?

Molodos, 2023-11-21:
I am sorry for the late response. Is this issue stiff present and if not, how did you fix it? (I am asking because I've seen, that you have already made a newer issue, which makes me think, that this problem could be fixed)
Best regards

Molodos, 2023-11-21:
I understand, thanks for that info. After some research, this seems to have something to do with your system configuration and not my script. As the issue is not that relevant, I will close it for now, but feel free to reopen it if there is updates :)

closed 2023-11-21 by Molodos