Molodos/ElegooNeptuneThumbnails-Prusa #10: Issue after update

DNicholai, 2023-11-21:
I performed misc program updates, and it broke the install!  

I'm getting the following error : 

```
./dist/ElegooNeptuneThumbnails-Prusa --printer=NEPTUNE4PRO /home/dpegon/Downloads/Shelf_Legs\ -\Kitch_Shelv_2023.11.21.stl 
Traceback (most recent call last):
  File "elegoo_neptune_thumbnails.py", line 249, in <module>
    thumbnail_generator: ElegooNeptuneThumbnails = ElegooNeptuneThumbnails()
  File "elegoo_neptune_thumbnails.py", line 25, in __init__
    self._thumbnail: QImage = self._get_q_image_thumbnail()
  File "elegoo_neptune_thumbnails.py", line 70, in _get_q_image_thumbnail
    base64_thumbnail: str = self._get_base64_thumbnail()
  File "elegoo_neptune_thumbnails.py", line 53, in _get_base64_thumbnail
    for line in file.read().splitlines():
  File "codecs.py", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe8 in position 80: invalid continuation byte
[65516] Failed to execute script 'elegoo_neptune_thumbnails' due to unhandled exception!

```

Any hints?

Molodos, 2023-11-21:
Hi,
it seems to have something to do with some special characters in the gcode before post processing. Could you maybe disable my thumbnails post processing script, generate the same gcode without it and send me that gcode file for further analysis? With that file, I will most probable be able to fix the error.
Thank you :)

DNicholai, 2023-11-21:
Nevermind, I think I ran the script (manually) on the saved PrusaSlicer file rather than the gcode, let's chalk this one up to user Error - Please close this issue :)

Molodos, 2023-11-21:
Happens to the best :)

closed 2023-11-21 by Molodos