Molodos/ElegooNeptuneThumbnails-Prusa #12: Neptune 4 Pro - Black thumbnail

Benna08, 2023-11-26:
Hi there, thanks for your great work with this script.

I followed all the steps and all the thumbnail previews are appearing as a black image on my Neptune 4 Pro. Any idea what would be causing this? 

Thank you so much.

Molodos, 2023-11-28:
Did you try using the "--printer=NEPTUNE4PRO" option to specify your printer manually? Also if you want to, you could send me one of the gcode files with the black thumbnails, maybe they could help me finding the error

Benna08, 2023-11-28:
Thanks for your reply.

Yes, I did specify the printer manually. Just to clarify, the whole
thumbnail isn't black. It shows the bed in grey and then the object in
black. It does it with all the gcodes I've tried. Over 20.

Benna08, 2023-11-28:
[image]

Benna08, 2023-11-28:
Here you go. :)

Molodos, 2023-11-28:
That looks like you have selected some black filament color in PrusaSlicer. I just know it from Cura: If you change the filament color in the slicer, it will also be like that in the thumbnail. Maybe you should look into that :)

Benna08, 2023-11-28:
Ah, that makes sense. Thank you so much. I'll give it a try right now. :)

Benna08, 2023-11-28:
Problem solved. Well, never really was a problem. Haha, I just had no Idea that the thumbnail preview actually represented the color of your filament also. Thanks for guiding me in the right direction and thanks again for this great script. :)

closed 2023-12-05 by Molodos