Molodos/ElegooNeptuneThumbnails-Prusa #15: Preview image issue.

Send66, 2024-01-05:
Hello, buddy. I would like to add a preview image to my own project, but you only have processed the encoded code here. I am unable to process the data of GImage and SImage on MCU. Can I release this encoding and decoding function because we cannot use your script? I would like to know how to convert compressed strings into RGB data. Thank you!

Send66, 2024-01-05:
I see_ Parse_ Thumbnail_ A new encoding compression has been applied to the new function, and I need to know the code for the decryption function, otherwise I won't be able to display the image on my screen.

Molodos, 2024-01-05:
Unfortunately, my Plugin only does the encoding, but maybe I can help if you give me mor information. Are you a 3D-printer manufacturer or something like that? In that case there might be a chance of cooperation in order to officially support your printers.

Send66, 2024-01-05:
Thank you for the author's early response to my question. No, I just wanted to use the image you encoded to display. Since you have re encoded it, I cannot know the RGB data inside. I see that there is a function for returning the encoding to its original in lib.dll. Can that be opened?

Send66, 2024-01-05:
Did you write the encoding function for converting RGB data to strings? I want to convert the string back to RGB raw data and display it on LCD, thank you!

Molodos, 2024-01-05:
Unfortunately I did not write the encoding and decoding myself, as I made the Plugin to work with Elegoo printers. But as you have already seen, there is a decoding function in ther original libraries. That is your only option if you want to add support for the plugin without a cooperation unfortunately.

Send66, 2024-01-05:
Isn't this original library written by you? Currently, I need to process image strings for RGB display, so I would like to inquire about it.

Send66, 2024-01-05:
Due to some commercial reasons, I am currently unable to disclose it. If it is because of this reason that the original function cannot be opened for encoding, I will communicate with the printer company later.

Molodos, 2024-01-05:
As already said, it was not written by another printer company. Unfortunately they did not want to provide me with details on the encoding, so good luck asking them, but I guess there is no chance.
As it looks like you would like your printer to be supported by my plugin (also the Cura version), the only thing I could offer you is a cooperation. In that way we could also agree on another encoding. But for that you needed to be not just a user with a single printer but a printer company, what you stated, you are not.

Molodos, 2024-01-05:
If there are things you don't want do discolse in this issue, I can offer you an email for a more private communication

Send66, 2024-01-05:
So I should know, it was written by a screen display design manufacturer. That's why we want to know, you know

Send66, 2024-01-05:
Can I give you your email address?

Molodos, 2024-01-05:
My email is: realmolodos@gmail.com

Send66, 2024-01-05:
Okay, I have sent the message over here.

closed 2024-01-05 by Molodos

201841301322, 2024-01-06:
I have successfully restored the str encoding to RGB data