Molodos/ElegooNeptuneThumbnails-Prusa #20: Next print's thumbnail becomes correct only after reboot

faruktoy, 2024-03-15:
It shows the correct thumbnail on the first print, but when I switch to a different print it shows the thumbnail of the first print. When I cancel printing and start the 2nd print again, the correct thumbnail. It seems like it can be fixed with the end gcode script.

Molodos, 2024-03-15:
If you ask me, that sounds like a caching issue within the printer firmware and has nothing to do with my script unfortunately.

faruktoy, 2024-03-15:
There is probably a problem with the situation you mentioned. What can I do? (Before using your script, I updated the firmware of my Neptune 3 pro printer and display to be [compatible.](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/blob/main/README-en.md) )

Molodos, 2024-03-17:
To be honest, I think there is not much you can do but waiting for a fix by Elegoo, unfortunately

faruktoy, 2024-03-17:
Thanks for your interest

closed 2024-03-17 by faruktoy