Molodos/ElegooNeptuneThumbnails-Prusa #21: error in the CMD-Windows (Win10), that opens when calling the plugin

schulahere, 2024-03-17:
Hi,
I call the plugin from the post processing in Orca (Windows 10).

In the window that opens I get the error (warning):

qt.gui.imageio: libpng warning: ICCP: known incorrect sRGB profile
QApplication::regClass: Registering window class 'QT652ScreenChangeObserverWindow' failed (Klasse ist bereits vorhanden) 

The picture in the gcode gets embedded however and is shown on the printer screen. 

Before I used the Version 2.1 (I think). With that Version the warning didn´t turn up.

Molodos, 2024-03-17:
In order to support the parameters in the corners, I had to switch the image framework. That's why the warning didn't occur earlier. However, it is just a warning that can be ignored. So as long as your thumbnails are displayed correctly, you're fine :D

closed 2024-03-17 by Molodos

schulahere, 2024-03-17:
o.k. I´ll ignore it. Thanks for the support!