Molodos/ElegooNeptuneThumbnails-Prusa #22: iCCP Error

tekgecko, 2024-03-17:
Running this on a Neptune 4 pro.

Output is set as follows:

C:\Users\samma\Documents\_Git\ElegooNeptuneThumbnails-Prusa\dist\ElegooNeptuneThumbnails-Prusa.exe --printer="NEPTUNE4PRO";

Getting this error and no thumbnail seen by the firmware:

[image]

Molodos, 2024-03-17:
The warnin should not cause any issues and can be ignored. However if the thumbnail isn't displayed, there must be another error.
If you share the generated gcode with me, I can have a look :)

tekgecko, 2024-03-17:
Sure, thank you! It seems to generate some odd stuff at the top of the file

https://1drv.ms/u/s!Ak9kyhzCNFc-g4JjDGPc69cjsO29dQ?e=eNZFch

tekgecko, 2024-03-17:
I tested further. Gcode from a network share doesn't display thumbnails on my Neptunes (I have several printers with a single mapped network share that hosts gcode). However, when I export the gcode to my desktop and transfer it directly to the printer using Fluidd, the thumbnails show up. This suggests the issue lies with how gcode is read from the network, not with your project.

Molodos, 2024-03-17:
Thanks! Actually, the "odd stuff" at the top is the encoded image in the pretty odd format, the Neptune printers are using. It looks perfectly fine to me.
Also, I tried it on my Elegoo Neptune 4 Pro and it shows just fine:
[image]

Molodos, 2024-03-17:
Okay, I see, that you found out the same thing, I guess that solves the issue :)

tekgecko, 2024-03-17:
Yes! Seems to be with the way the firmware reads from the network, I will investigate further, thanks for your help!

closed 2024-03-17 by tekgecko