Molodos/ElegooNeptuneThumbnails-Prusa #23: Not able to see icon on Neptune4 control screen

Keyplayermark, 2024-03-22:
I followed the instructions to the letter. It does create an icon in the windows directory.  However; there is no icon on the my Elegoo UI screen.
I tried to troubleshoot, but I'm a retired Network and Communications engineer, the only language I know is Basic a touch of C++ and old 8088 Assembly.  LOL.

Prusa slicer version 2.7.2
Neptune 4 UI is 1.1.5
Neptune 4 Firmware 1.92
Windows 10 pro.

qt.gui.imageio:  libpng warning: iCCP: known incorrect sRGB profile
QApplication::regClass:  Registering window class 'Qt652ScreenChangeObserverWindow' failed.  (Class already exists.)

Molodos, 2024-03-25:
Could you maybe share the generated gcode file with me. By looking into it, I will be able to understand the situation better :)

Keyplayermark, 2024-03-25:
Greetings; Thank you for your response.
So, it turns out that If I print through the Moonraker interface via LAN,
the icons are shown on the Klipper Pad interface.  However; If I save the
same file onto a USB and print from it, there is no icon picture.
For me, this is fine because I rarely print from a USB.

So it appears to be a thing with Klipper and the interface I guess.

ciao.

Molodos, 2024-03-25:
Very interesting. Seems to be a bug in the printer firmware. Anyways thanks for getting back to me with that info :)

closed 2024-03-25 by Molodos