Molodos/ElegooNeptuneThumbnails-Prusa #25: Error when adding a filament change

schulahere, 2024-03-31:
When exporting a g-code file (calling the post processing script) that includes a filament change in orca the script terminates with an error and the file doesn´t get exported or transferred to the printer.
Everything works as expected when using the same file without the filament change command.

[image]

[image]

Molodos, 2024-04-01:
Thanks for reporting and including all the details! I just published a fix :)

closed 2024-04-01 by Molodos

schulahere, 2024-04-03:
thx, I´ll try it...